k3ng/k3ng_rotator_controller #23: Auto-park feature

somewhere-or-other, 2016-03-25:
What would you think about an auto-park feature, where the controller, if it  doesn't receive a command for a configurable amount of time (eg. 15 minutes), moves the rotator into a specific "parking" position?

The idea here is if you live in a area with strong-but-intermittent wind, you may want to park your rotator in a specific position, to minimize the wind load on an antenna mast/tower/etc.  Having it auto-park after a timeout, would make it more reliable, and avoid needing to manually park it each time.

As I come up to speed on this code base, I'll see if I can build it, and send you a pull request.  I'm just getting started, though, so it might be a little while.

Lloyd  - KC7NGB

k3ng, 2016-03-26:
Hi Loyd.  It sounds like a good feature to me.  I'm working on a big
feature for the keyer project and I have an update queued for the tuner
project, so it might be awhile for me to get to this as well.  Be we can
definitely do it, so it's on the list.

ON7KGK, 2017-07-24:
Hello K3NG,

Your project is really great ! 👍
All the features I have try as rtc, gps, i2c display, joystick worked perfectly.

Just one question :  Is the auto park feature (after 15 min) is implemented now ? I don't find it in the documentation.
My remote qth is really windy and I should park the array antennas in the prevalling winds during no activity.  I'm newbie in programming code.

I know you have another project on the table but do you think this feature is easy to add?
Thank you

Best 73's from Belgium. 
Michaël ON7KGK

k3ng, 2017-08-03:
I am working on this feature.  Should the autopark timer be reset only when
the antenna is rotated, or should it also be reset if a serial command is
entered (any command, including ones that don't initiate rotation)?

k3ng, 2017-08-03:
Updated code uploaded with new FEATURE_AUTOPARK.

Documented here:
https://github.com/k3ng/k3ng_rotator_controller/wiki/705-Park-and-AutoPark

Please test.

73

ON7KGK, 2017-08-03:
Hello.

Thank you for the code. I will test it this week-end. I think that for my usage the timer should only be reset when the antenna is rotated but I am not sure because I will use pstrotator for remote work and I don't know if the commands from this software are serial or not.

The autopark timer (in minute) must be entered just one time or every time the rotor is power on ?
I never used the console to send serial commands.

Perhaps, setting the timer in the features file should be easier for me. For example : #define AUTOPARK_TIMER 15

My activity is mainly DATV so the antenna can stay in same direction for a long time qso (1h or more).

So is it possible to enable and disable the autopark feature with a pin ?

Have a nice day.
Thank you

Michaël from Belgium.

k3ng, 2017-08-15:
The autopark timer setting is written to EEPROM, so you only need to set it
once.

How about I create two pins:

pin_autopark_disable: Pull low to disable autopark

pin_autopark_timer_reset: Pull low to reset the autopark timer (tie in with
rig PTT)

ON7KGK, 2017-08-15:
Hello,

Nice !  You can do like that.  Perfect for me
Big thank you

Michaël

k3ng, 2017-09-03:
Uploaded updated code to Github, version 2017.09.03.02:

      Added pins pin_autopark_disable and pin_autopark_timer_reset for
FEATURE_AUTOPARK

Please test!

73
Goody
K3NG