k3ng/k3ng_rotator_controller #26: Code not compiling

VBColdie, 2016-06-28:
I would have thought this code by now would have been used and compiled by many users, yet all I get is a screen full of errors, this is the screen dump, can anyone help with why?
Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from sketch\libraries\TimerFive\TimerFive.h:16:0,

                 from sketch\libraries\TimerFive\TimerFive.cpp:16:

sketch\libraries\TimerFive\TimerFive.cpp: In function 'void TIMER5_OVF_vect()':

sketch\libraries\TimerFive\TimerFive.cpp:20:5: warning: 'TIMER5_OVF_vect' appears to be a misspelled signal handler [enabled by default]

 ISR(TIMER5_OVF_vect)          // interrupt service routine that wraps a user defined function supplied by attachInterrupt

     ^

sketch\libraries\TimerFive\TimerFive.cpp: In member function 'void TimerFive::initialize(long int)':

sketch\libraries\TimerFive\TimerFive.cpp:27:3: error: 'TCCR5A' was not declared in this scope

   TCCR5A = 0;                 // clear control register A 

   ^

sketch\libraries\TimerFive\TimerFive.cpp:28:3: error: 'TCCR5B' was not declared in this scope

   TCCR5B = _BV(WGM53);        // set mode as phase and frequency correct pwm, stop the timer

   ^

In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,

                 from sketch\libraries\TimerFive\TimerFive.h:15,

                 from sketch\libraries\TimerFive\TimerFive.cpp:16:

sketch\libraries\TimerFive\TimerFive.cpp:28:16: error: 'WGM53' was not declared in this scope

   TCCR5B = _BV(WGM53);        // set mode as phase and frequency correct pwm, stop the timer

                ^

sketch\libraries\TimerFive\TimerFive.cpp: In member function 'void TimerFive::setPeriod(long int)':

sketch\libraries\TimerFive\TimerFive.cpp:35:62: error: 'CS50' was not declared in this scope

   if(cycles < RESOLUTION)              clockSelectBits = _BV(CS50);              // no prescale, full xtal

                                                              ^

sketch\libraries\TimerFive\TimerFive.cpp:36:62: error: 'CS51' was not declared in this scope

   else if((cycles >>= 3) < RESOLUTION) clockSelectBits = _BV(CS51);              // prescale by /8

                                                              ^

sketch\libraries\TimerFive\TimerFive.cpp:37:62: error: 'CS51' was not declared in this scope

   else if((cycles >>= 3) < RESOLUTION) clockSelectBits = _BV(CS51) | _BV(CS50);  // prescale by /64

                                                              ^

sketch\libraries\TimerFive\TimerFive.cpp:37:74: error: 'CS50' was not declared in this scope

   else if((cycles >>= 3) < RESOLUTION) clockSelectBits = _BV(CS51) | _BV(CS50);  // prescale by /64

                                                                          ^

sketch\libraries\TimerFive\TimerFive.cpp:38:62: error: 'CS52' was not declared in this scope

   else if((cycles >>= 2) < RESOLUTION) clockSelectBits = _BV(CS52);              // prescale by /256
[... 318 lines cut ...]

   TCCR5B &= ~(_BV(CS50) | _BV(CS51) | _BV(CS52));          // clears all clock selects bits

                               ^

sketch\libraries\TimerFive\TimerFive.cpp:97:43: error: 'CS52' was not declared in this scope

   TCCR5B &= ~(_BV(CS50) | _BV(CS51) | _BV(CS52));          // clears all clock selects bits

                                           ^

sketch\libraries\TimerFive\TimerFive.cpp: In member function 'void TimerFive::restart()':

sketch\libraries\TimerFive\TimerFive.cpp:102:3: error: 'TCNT5' was not declared in this scope

   TCNT5 = 0;

   ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

k3ng, 2016-06-28:
Remove the TimerFive files from your ino directory and recompile.  The Uno
doesn't support Timer Five.

VBColdie, 2016-06-28:
Thankyou for your very quick response.  Can you elaborate further please on
how to do that, I'm very new to this coding stuff, but I need to know do
you mean to delete the directory with those files, or is there a routine
calling those files in the code? I assume it's in the code to call the
library, but I can't see where.

k3ng, 2016-06-28:
Remove TimerFive.cpp and TimerFive.h from the directory that contains the
k3ng_rotator_controller.ino file.

VBColdie, 2016-06-29:
[image]

VBColdie, 2016-06-29:
Where in directory is files? I see no reference to TimerFive !

k3ng, 2016-06-29:
Remove the libraries directory from this directory.  Place the files that
are within that libraries directory in your \sketchbook\libraries\
directory.

VBColdie, 2016-06-29:
You're a genius, code compiled perfectly, thanking you sincerely.

k3ng, 2016-06-29:
Not a genius, just been doing it for a little bit.  Have fun and enjoy the
code.  73 :-)

closed 2016-09-08 by k3ng