k3ng/k3ng_rotator_controller #27: Runtime Error 8080

VBColdie, 2016-07-02:
Every time I initiate rotor control I get "Runtime error 8080 Comm port opening error" Yes I'm sharing the comm port with the Arduino, but your instructions say this is ok, the port is set to 9600 and "Easycomm" protocol is selected. The Arduino blinking led for the comm port activity flashes perfectly but then I initiate a move and I get the error and the program crashes every time. Any ideas please? It was somewhat more stable before I initiated AZ/EL but that's what I need, and the project is in breadboard stage but otherwise stable.

k3ng, 2016-07-02:
I don't know what program you're using that is giving the runtime error and
I can't really provide support for programs anyway.  You can't share com
ports between programs or multiple pieces of hardware.

VBColdie, 2016-07-02:
Hi, I'm trying to setup PSTRotator and setting it for Easycom as K3NG(easycom) in the setup menu. I thought I could consider the Arduino USB port (currently 12) as being the same port for output as per PSTRotator settings. So you're saying that I can't breadboard the whole setup on the same comm port?

VBColdie, 2016-07-02:
I quote this paragraph from your instruction page 310.
"The unit is interfaced to the computer simply using the native Arduino USB port. Currently Yaesu and Easycom protocols are supported."

k3ng, 2016-07-02:
OK, if you mean you can use the Arduino port for programming and
interfacing to control programs, yes, that's the way it works.  If another
program (i.e. the Arduino IDE) is connected to the Arduino, you need to
exit that program so PST can grab the COM port.  You can't have two
programs using a COM port.

k3ng, 2016-07-02:
> Oh ok I'll try closing the Arduino IDE and see what happens, thanks
> again....stay tuned.

OK good luck

closed 2016-09-08 by k3ng