k3ng/k3ng_rotator_controller #28: Using 4_BIT_DISPLAY

VBColdie, 2016-07-03:


VBColdie, 2016-07-03:
When uncommenting FEATURE_4_BIT_LCD_DISPLAY in rotator_features.h to make my display work I then get a dump of errors.  I've been ignoring them but I want an error free compile, see attached the dump, can you have a look please Anthony.

C:\Users\COLINC~1\AppData\Local\Temp\arduino_modified_sketch_357901\k3ng_rotator_controller.ino: In function 'char* idle_status()':

C:\Users\COLINC~1\AppData\Local\Temp\arduino_modified_sketch_357901\k3ng_rotator_controller.ino:3289:12: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   return("");

            ^

C:\Users\COLINC~1\AppData\Local\Temp\arduino_modified_sketch_357901\k3ng_rotator_controller.ino: In function 'void initialize_display()':

C:\Users\COLINC~1\AppData\Local\Temp\arduino_modified_sketch_357901\k3ng_rotator_controller.ino:6696:163: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

       k3ngdisplay.print_center_timed_message("\x4B\x33\x4E\x47","\x52\x6F\x74\x6F\x72\x20\x43\x6F\x6E\x74\x72\x6F\x6C\x6C\x65\x72",CODE_VERSION,SPLASH_SCREEN_TIME);

                                                                                                                                                                   ^

C:\Users\COLINC~1\AppData\Local\Temp\arduino_modified_sketch_357901\k3ng_rotator_controller.ino:6696:163: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\COLINC~1\AppData\Local\Temp\arduino_modified_sketch_357901\k3ng_rotator_controller.ino:6696:163: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

Sketch uses 28,912 bytes (89%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,921 bytes (93%) of dynamic memory, leaving 127 bytes for local variables. Maximum is 2,048 bytes.
Low memory available, stability problems may occur.

k3ng, 2016-07-04:
You can ignore compiler warnings.  They won't affect the performance or
operation of the keyer.  However, the compiler noted you're low on local
variable memory.  As it mentions, you may have stability problems (i.e.
crashes, unexpected stuff happening).  You should reduce the feature set or
upsize to a Mega.

VBColdie, 2016-07-05:
Anthony, firsty let me say that I really appreciate your help and feedback
to my questions. But I need to bring to your attention a problem that I
haven't explained properly. When I compile with 4_BIT_LCD_DISPLAY
uncommented in "rotator_features.h" I get the the following:
Sketch uses 25,596 bytes (79%) of program storage space. Maximum is 32,256
bytes.
Global variables use 1,351 bytes (65%) of dynamic memory, leaving 697 bytes
for local variables. Maximum is 2,048 bytes.
 No errors, no problems,
But as for the dump I sent you before, the only change is to comment that
feature, and is the ONLY change made to the file, yet that's when I get the
error messages about the lack of memory. So just letting you know it goes
beyond not only the compile errors, it then sucks the memory up! Maybe you
could simulate what I'm saying and see for yourself.
Thank again,
Colin.

k3ng, 2016-07-05:
So with 4_BIT_LCD_DISPLAY uncommented you get the error message about
memory, and with it commented out you don't?

VBColdie, 2016-07-05:
Exactly mate, and it is only a matter of // on that one line !

k3ng, 2016-07-05:
That's a feature, not a bug :-)  When you uncomment that line, a whole
bunch more code gets compiled.  More functionality, more code, less memory
free....  (The compiler warnings aren't chewing up the memory)

VBColdie, 2016-07-05:
So please clarify your statement " (The compiler warnings aren't chewing up
the memory) "  I don't understand mate. Are you saying that one should
upgrade from a UNO to a MEGA just to support that extra functionality?
I've ordered a later model display with buttons incorporated so I'll wait
to see what happens with it before I get to worried, because I'm hoping
that it will work without having to uncomment that line and see what
happens.  For your information I have a long development road ahead of me
before I get my project sorted and i'll be inevitably bugging you now and
again, but rest assured when I've finished my final design I'm sure
you'll appreciate my input (hopefully), but please don't be upset with me
for bugging you, everyone needs someone to turn to for support when needed,
and who better to turn to than the author :) Thanks again Anthony.

Sent with MailTrack
<https://mailtrack.io/install?source=signature&lang=en&referral=colin.coldie@gmail.com&idSignature=22>

k3ng, 2016-07-05:
Yes.  In general I recommend using a Mega.  It's hard to stuff all the
functionality into an Uno.

somewhere-or-other, 2016-07-05:
I don't know if it will be helpful or not, but you can look at my fork here as an example of fitting into a smaller memory footprint; I'm able to get most of the functionality to fit in a cheap, Uno SMD clone which has the smaller memory footprint: https://github.com/somewhere-or-other/k3ng_rotator_controller

If you do use that one as a starting place, be aware that I'm using an I2C, 4x20 char display, among other things.

When you're dealing with the smaller memory footprint, it is possible to make it work, but it's a bit of a balancing act.  For example, if I want to add the debugging code, or the GPS code, or the sun/moon tracking code, I'd have to eliminate something to make room.

Anthony is right.  If you're doing something more difficult, needs more functionality, etc., you'd be better off to get something with a bigger memory footprint like a Mega or a Due.

VBColdie, 2016-07-05:
Thanks heaps mate for the info, I've only just begun my Arduino learning
curve and any info I can get helps, cheers.

somewhere-or-other, 2016-07-05:
Note that I edited my earlier comment, to point to a specific branch. Originally I just pointed to the master branch, which is in sync with Anthony's master.  Sorry for the confusion, but if you're just going off the emailed link, it might not be that helpful.

VBColdie, 2016-07-05:
Yes now I'm confused (can't see any edit) could you post me the link please.

somewhere-or-other, 2016-07-06:
https://github.com/somewhere-or-other/k3ng_rotator_controller/tree/n7byu_azel_settings

VBColdie, 2016-07-09:
Hi again, can you tell me why I get the following error when trying to
compile your version of this software when you can clearly see the file in
the directory tree (see attached screen shot) please.

Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files
(x86)\Arduino\k3ng_rotator_controller-n7byu_azel_settings\k3ng_rotator_controller\k3ng_rotator_controller.ino:504:30:
fatal error: rotator_hardware.h: No such file or directory

 #include "rotator_hardware.h"

                              ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

somewhere-or-other, 2016-07-09:
I have no idea. Are both the "rotator_features.h" and the "k3ng_rotator_controller.ino" files still there in the "C:\Program Files (x86)\Arduino\k3ng_rotator_controller-n7byu_azel_settings\k3ng_rotator_controller\" directory?

VBColdie, 2016-07-09:
The program forces ("k3ng_rotator_controller.ino") into a subdirectory in
the file tree, tried taking it out of the subdirectory but it forces it
back into a subdirectory!

Sent with MailTrack
<https://mailtrack.io/install?source=signature&lang=en&referral=colin.coldie@gmail.com&idSignature=22>

closed 2016-09-08 by k3ng