k3ng/k3ng_rotator_controller #30: incorrect index for row_override

dfannin, 2016-08-30:
There may be an issue with the LCD display code logic, which I found when setting the  "LCDxxxROW" and "OPTION_DISPLAYxxx" defines. Would like to see it confirmed before sending a patch.

The issue is that in the main .ino file, the variable  "row_override[]" array is indexed incorrectly ; It's off by one.  

Here's the problem:

1) row_override is initialized as "byte row_override[LCD_ROWS]"  
   LCD_ROWS is the number of ROWS (so its indexed starting at 1), and all elements of the array are set to 0 (false).  However, row_override is an array with index 0.  if LCD_ROWS=4 , then row_override would have keys from 0-3. 
2)   it is then referenced to determine if the row should be printed. ex:
    if( ! row_override[LCD_BIG_CLOCK_ROW]) ... 
    but its using the LCD_xxx value, which is set to the ROW, which is index 1. 
3) so this check is not working correctly for setting things like LCD_BIG_CLOCK_ROW to 4.  In this case, it would resolve to row_override[4], which means an array index out of bounds. 

The suggested fix would be to modify all the "row_override[LCD_xxx]" references to "row_override[LCD_xxx - 1]" . That should fix the misaligned indexes.

Due to the error, this issue occurs only when you using something like row 4, so it sometimes causes a issue and sometimes not.

k3ng, 2016-08-30:
Hi David.  You are correct.  Thanks for catching this.  Please submit.  73
Goody

dfannin, 2016-08-30:
Thanks, Closing this one out.

closed 2016-08-30 by dfannin