k3ng/k3ng_rotator_controller #39: POLOLU_LSM_303_MIN_ARRAY

canislupus11, 2017-09-19:
When i put 
#define POLOLU_LSM_303_MIN_ARRAY {-822, -644, -1151}
to rotor_settings.h, i get strange behavior of the controller. I set the azimuth, the controller starts rotating the rotor, but when the target is reached does not stop. Elevation is ok. If I do not have this line, everything is ok.

canislupus11, 2017-11-12:
:-(

k3ng, 2017-11-13:
Sorry, I've been busy moving my home.  I'm very behind on support requests.

ba5dx, 2019-01-30:
Reduce the rotation speed of the LSM303 and try to put it in the center position

aka9950202, 2019-02-19:
I am having a similar problem. Using the calibrating data helped to a point.

When passing 359 it jumps to somewhere in the 20's and never stops until it runs to the mechanical stop.

I tried changing the centering from N to S did not help.

I did notice the screen goes blank before the run away happens.