k3ng/k3ng_rotator_controller #40: Response to Yaesu GS232 command "C" on overlap

dh1tw, 2017-10-04:
Hey,

this question is not directly related to this software, but I hope somebody could help me out:

I'm currently writing a [web frontend for rotators](https://github.com/dh1tw/remoteRotator). Currently, the Yaesu G232 protocol is supported. However I don't have a yaesu rotator available here. 

The command ```C``` and ```C2``` allow to query the heading of the rotator. Can somebody confirm me the range of the response, in particular when the rotator is overlapping. Does the rotator return ```0...359``` or ```0...449``` ?

It would be useful to know in order to implement the visualisation of Overlap properly.

Any help appreciated!

73 Tobias
DH1TW

[image]

k3ng, 2017-10-06:
I believe the standard behavior of the Yaesu unit is to respond with 0 to
359, not 0 to 450.  I can certainly add an option to the Arduino rotator
code to make it respond with 0 to 450, but that would naturally work only
with this project and not other implementations.

dh1tw, 2017-10-08:
Hey Anthony,
thanks for the clarification. That helps. I think I found a way to detect overlap with the metadata the user can provide through cli flags. I prefer to be standard compliant :-)

BR Tobias

closed 2017-10-08 by dh1tw