k3ng/k3ng_rotator_controller #46: Head problem on 16x2 LCD display

maxmiztejm, 2018-02-20:
Hello K3NG!
I trying make an your controller with 16x2 Display.
The head is shown in this way:
Azimuth XXX.X
  zimuth XXX.X
The XXX.X are same in both rows. AZ/EL control feature is enabled. Problem is, that is no Elevation XXX.X text on the second row. Do you have any idea, where is the problem? Thank you!

k3ng, 2018-02-21:
What options and settings do you have configured?

maxmiztejm, 2018-02-21:
#define OPTION_DISPLAY_STATUS
#define OPTION_DISPLAY_HEADING
#define OPTION_DISPLAY_HEADING_AZ_ONLY
#define OPTION_DISPLAY_HEADING_EL_ONLY

#define FEATURE_4_BIT_LCD_DISPLAY
#define LCD_COLUMNS 16 
#define LCD_ROWS 2

k3ng, 2018-02-21:
I also need to see the settings you have for the various rows, like these:

#define LCD_HEADING_ROW 2
#define LCD_HEADING_FIELD_SIZE 20
#define LCD_AZ_ONLY_HEADING_ROW 1
#define LCD_AZ_ONLY_HEADING_FIELD_SIZE 20
#define LCD_EL_ONLY_HEADING_ROW 2
#define LCD_EL_ONLY_HEADING_FIELD_SIZE 20
#define LCD_STATUS_ROW 1
#define LCD_STATUS_FIELD_SIZE 20

However, not knowing these, I would say turn off OPTION_DISPLAY_HEADING
because it's likely it's interfering with OPTION_DISPLAY_HEADING_EL_ONLY if
you're using the default settings.

maxmiztejm, 2018-02-21:
I already, try this. But no text: Elevation XXX.X is displayed.

k3ng, 2018-02-21:
Do this:

#define OPTION_DISPLAY_STATUS
#define OPTION_DISPLAY_HEADING
// #define OPTION_DISPLAY_HEADING_AZ_ONLY
// #define OPTION_DISPLAY_HEADING_EL_ONLY

Or this:

// #define OPTION_DISPLAY_STATUS
// #define OPTION_DISPLAY_HEADING
#define OPTION_DISPLAY_HEADING_AZ_ONLY
#define OPTION_DISPLAY_HEADING_EL_ONLY

You have only two rows to work with, so there are conflicts occurring.

maxmiztejm, 2018-02-21:
First option: displayed Azimuth XXX.X on the second row.
Second option: displayed Azimuth XXX.X on the first row.

maxmiztejm, 2018-02-21:
Mine settings
#define LCD_COLUMNS 16 // Zmena! //20 
#define LCD_ROWS 2 // Zmena! //2       // this is automatically set below for HARDWARE_EA4TX_ARS_USB and HARDWARE_M0UPU
#define LCD_UPDATE_TIME 1000           // LCD update time in milliseconds
#define LCD_HHMM_CLOCK_POSITION LEFT          //LEFT or RIGHT
#define LCD_HHMMSS_CLOCK_POSITION LEFT          //LEFT or RIGHT
#define LCD_ALT_HHMM_CLOCK_AND_MAIDENHEAD_POSITION LEFT
#define LCD_ALT_HHMM_CLOCK_AND_MAIDENHEAD_ROW 1
#define LCD_CONSTANT_HHMMSS_CLOCK_AND_MAIDENHEAD_POSITION CENTER
#define LCD_CONSTANT_HHMMSS_CLOCK_AND_MAIDENHEAD_ROW 3
#define LCD_BIG_CLOCK_ROW 4
#define LCD_GPS_INDICATOR_POSITION RIGHT //LEFT or RIGHT
#define LCD_GPS_INDICATOR_ROW 1
#define LCD_MOON_TRACKING_ROW 3                                   // LCD display row for OPTION_DISPLAY_MOON_TRACKING_CONTINUOUSLY
#define LCD_MOON_TRACKING_UPDATE_INTERVAL 5000
#define LCD_SUN_TRACKING_ROW 4                                    // LCD display row for OPTION_DISPLAY_SUN_TRACKING_CONTINUOUSLY
#define LCD_SUN_TRACKING_UPDATE_INTERVAL 5000
#define LCD_MOON_OR_SUN_TRACKING_CONDITIONAL_ROW 3                // LCD display row for OPTION_DISPLAY_MOON_OR_SUN_TRACKING_CONDITIONAL
#define SPLASH_SCREEN_TIME 3000

#define LCD_HEADING_ROW 2
#define LCD_HEADING_FIELD_SIZE 20
#define LCD_AZ_ONLY_HEADING_ROW 1
#define LCD_AZ_ONLY_HEADING_FIELD_SIZE 20
#define LCD_EL_ONLY_HEADING_ROW 2
#define LCD_EL_ONLY_HEADING_FIELD_SIZE 20
#define LCD_STATUS_ROW 1
#define LCD_STATUS_FIELD_SIZE 20
#define LCD_DIRECTION_ROW 1
#define LCD_HHMMSS_CLOCK_ROW 1
#define LCD_HHMM_CLOCK_ROW 1
#define PARKING_STATUS_DISPLAY_TIME_MS 5000

k3ng, 2018-02-21:
How much local variable space do you have available (shown at compile time)?

maxmiztejm, 2018-02-21:
75% are available

k3ng, 2018-02-21:
How many bytes for local variables?

maxmiztejm, 2018-02-21:
2105 bytes

maxmiztejm, 2018-02-21:
leaving 6087 bytes for local variables

maxmiztejm, 2018-02-21:
Problem solved. 
I forget change FEATURE_ELEVATION_CONTROL setting. Now it works. Thank you!

k3ng, 2018-02-22:
Ah, OK.  I'm going to add some code to prevent this in the future for
others.

73
Goody
K3NG