k3ng/k3ng_rotator_controller #47: complete schematics

yo8sxt, 2018-02-26:
hello,i wanth to make this project but i have difficulties.please maybe you can give a complete and up to date electrical schematics and the program that works whit all the pieces from the schematics.i am new in this but i wanth to push my limits so please  help me in this...73

yvest, 2018-02-26:
Hello,

There is a lot of documentation on the github site:
https://github.com/k3ng/k3ng_rotator_controller/wiki/210-Hardware-Schematic
[https://avatars0.githubusercontent.com/u/3332720?s=400&v=4]<https://github.com/k3ng/k3ng_rotator_controller/wiki/210-Hardware-Schematic>

210 Hardware Schematic · k3ng/k3ng_rotator_controller Wiki ...<https://github.com/k3ng/k3ng_rotator_controller/wiki/210-Hardware-Schematic>
k3ng_rotator_controller - K3NG Arduino Amateur Radio Rotator Controller
github.com

And on K3NG's web site: https://blog.radioartisan.com/yaesu-rotator-computer-serial-interface/
[https://radioartisan.files.wordpress.com/2011/03/k3ng_arduino_rotator_controller_2011092801.png?w=300]<https://blog.radioartisan.com/yaesu-rotator-computer-serial-interface/>

Arduino Rotator Computer Interface and Controller | Radio ...<https://blog.radioartisan.com/yaesu-rotator-computer-serial-interface/>
Updated 2018-02-01 Introduction This is an Arduino-based rotator interface that interfaces a computer to a rotator or rotator controller, emulating the Yaesu GS-232A ...
blog.radioartisan.com

Their are kits available, such as this one:
https://remoteqth.com/single-rotator-interface.php
RemoteQTH Single rotator interface<https://remoteqth.com/single-rotator-interface.php>
Single USB azimuth rotator interface - three-wire potentiometer input (voltage divider) This is part of Multi rotator control
remoteqth.com

Are you familiar with Arduino? It's the heart of the system and you will need to learn how to use it in order to configure and program your controller.

Regards,

-yvest

k3ng, 2018-02-27:
Thanks, yvest

YO8SXT, I recommend you join the Radio Artisan group for support https://groups.io/g/radioartisan

73
Goody
K3NG

closed 2018-02-27 by k3ng