k3ng/k3ng_rotator_controller #49: 16x2 I2C lcd not working

kimtheis, 2018-04-10:
Hi my 16x2 i2c lcd is not showing any thing

k3ng, 2018-04-10:
Please post support requests on https://groups.io/g/radioartisan .  Thanks

AngelSerra, 2019-07-05:
i,m used 
in file rotator_features.ini
#define FEATURE_RFROBOT_I2C_DISPLAY

and file rotator_ k3ngdisplay.h

#define FEATURE_RFROBOT_I2C_DISPLAY

df7cb, 2022-10-23:
Fwiw, that the display has to be enabled in *two* places has also cost me some hair pulling. 

rotator_features.h: `#define FEATURE_4_BIT_LCD_DISPLAY`
rotator_k3ngdisplay.h: `#define FEATURE_4_BIT_LCD_DISPLAY`

The fact it's already enabled in k3ngdisplay.h by default didn't make spotting the other place easier.

But I got it working... Thanks! :)