k3ng/k3ng_rotator_controller #54: Shows two lines only with the word "Azimuth º"

DirceuPY5DK, 2018-09-12:
Hello everyone.
I am using a MEGA 2560, incremental encoder CWZ6C Omron, tinyGPS Neo6M and 4x20 display ... it occurs that on the screen appear two lines written "Azimuth º" but does not appear the direction of the antenna ... however when turning preset encoder appears the target angle. Please, can someone help me why the display shows the two lines without the data ... and how do I add lat / long on the fourth line? Thank you. PY5DK
[image]
[image]

MrTom1, 2018-12-28:
Hi,
I have similar problem. After upgrading to latest software release (using same configuration parameters) controller is still working fine but display not showing azimuth anymore. Any idea?

DG7NT
[image]

DirceuPY5DK, 2023-06-16:
Tnx. @MrTom1

 All ok now.

Best regards,