k3ng/k3ng_rotator_controller #65: Which version is Ok for Arduino UNO ?

scandroale, 2020-04-12:
Hi my friend good work.. i try to use this project with arduino Uno .. but the memory is 101%
please tell me who is the last version compatible whit arduino uno i have 3 arduino uno.
I have read this https://groups.io/g/radioartisan/topic/25512501#8749 and an user tell me the version of 2013.
you have a lite version of this project updated ?
i use a LCD simple 16 X 2
Thanks a lot 73 IW2KVT

lu9cbl, 2020-04-12:
in our case with Az and Elevation feature enabled, AND the display, we need to migrate to Arduino MEGA, because the Arduino One dont have enough memory... The pinout it's the same, so we reuse the same shield

scandroale, 2020-04-12:
Thanks there isn't a lite version for arduino Uno ?? i need test it, i have need locate and chase the satellites with H-V elevation

Krytic, 2020-04-13:
I've used this script on an UNO before. I had to delete all the comments, and remove definitions for unused features, but it can in principle be done.

G6ZRV, 2020-04-18:
scandroale Did you find a version that will run on the Uno ?? i would like to use my Uno for the rotator but cant find a version of K3NG that will run.

n1iic, 2021-04-09:
@Krytic Sean, by chance can you publish the UNO version? I am a beginner with arduino and have a slim chance of having it work. Thank you!

G6ZRV, 2022-08-31:
> I'm using last k3ng code with an arduino Uno to control my Yaesu G-5500 (az+el).
> More info here https://www.ea5wa.com/arduino/controlador-rotor-g-5500

Hi Juan, yes I did find a code that works for me I have built it and it is working great.
Here is a link to a YouTube video it’s in 3 parts. If you have questions about it email Virol and he will advise you on your question.
https://youtu.be/VlOZKF5xgn0
73
Peter G6ZRV