k3ng/k3ng_rotator_controller #68: Yaesu GS232 Commands when using Remote and Slave

VK5LO, 2020-04-26:
Hi I have been trying to workout how the calibration for AZ and Elevation is done on a remote slave unit when sending commands though a master.  When talking to the serial port on the master, I can send all commands, B, C, C2, S, A, E, L, R, D, U,  and these all work on the slave, but calibration commands O, F, O2, F2 do not work, and there is no data sent back other than a ?>  I have gone through the code many times and cannot see why these calibration functions won't work.  Could this be a bug in the software that has not come up before?

k3ng, 2020-04-26:
What kind of sensors are you using?

VK5LO, 2020-04-26:
Sorry I forgot to mention they are pots connected to the slave unit.  They are being read OK and showing up on the display on the master.  Maybe its something to do with where the calibration settings are stored and if using sensors on the slave unit cannot be stored?

k3ng, 2020-04-26:
The calibration is stored on a slave unit.

Are you sending the commands from the master using the \S command, or
connecting your computer to the remote unit control port?

VK5LO, 2020-04-26:
I am using the computer with arduino ide serial monitor on the master sending the yaesu serial commands to control it.  All work except the calibration commands.  I though that maybe I could do it on the remote slave but it just gives back I think it says ERR2

k3ng, 2020-04-26:
OK, let me look at the code a bit.  Maybe I can rig this up on the bench
and test.

In the meantime, can you connect the remote unit directly to your computer
and see if the O and F commands work?

VK5LO, 2020-04-26:
The master on its own when not defined, ( uncommented ) as a FEATURE_MASTER_WITH_SERIAL_SLAVE works fine its only when this option is selected and  I can send it O and F commands it works fine.

k3ng, 2020-04-28:
On the master (with FEATURE_MASTER_WITH_SERIAL_SLAVE enabled) and with the
slave connected, could you execute on the master:

\Z
\R
\SO

Does this show you the "rotate fully CCW" message?

VK5LO, 2020-04-28:
OK here are the message replies Anthony
\Z  Suspend auto remote commands on
\R Remote port rx sniff on
\SO ER02

VK5LO, 2020-04-29:
I think that the issue is to do with #ifdef FEATURE_AZ_POSITION_POTENTIOMETER  somewhere in the code around line 14192 latest version.  In the master of course its set to #define FEATURE_AZ_POSITION_GET_FROM_REMOTE_UNIT and in the remote unit set to #define FEATURE_AZ_POSITION_POTENTIOMETER