k3ng/k3ng_rotator_controller #91: Accurate Calibration

IW8ELN-4U, 2022-06-26:
I cannot understand how to perform the calibration using the parameters:

#define AZIMUTH_CALIBRATION_FROM_ARRAY {180,630}
#define AZIMUTH_CALIBRATION_TO_ARRAY {180,630}

I need these parameters because my rotor has no linear variation, so I would like to create more measurement points to make the AZ reading accurate.

With the debub I can read the "Analog" value and therefore I am able to attribute the right "Analog" value to each position.
In my case Analog is between 3 and 517 (3 = position -180 and 517 +180 with central North).
I could not find explanations regarding the values indicated, I only understood that I have to create multiple FROM_TO but it is not clear to me how to enter the parameters inside.
You can give me an example with:

Position AZ 90 ° = Analog 300
Position AZ 270 = Analog 175
Position AZ 180 = Analog 225

(for example)

The limit switches are perfect (CCW and CW) made via the terminal.
73 de IW8ELN Roberto

Philips76, 2022-09-18:
Ciao, hai risolto?
'73  IU7PTJ(IZ7DDA)

IW8ELN-4U, 2022-09-20:
> Ciao, hai risolto? '73 IU7PTJ(IZ7DDA)

Si, trovi la soluzione nel gruppo Remote.

Philips76, 2022-09-20:
Si grazie

#define AZIMUTH_CALIBRATION_FROM_ARRAY \
  { 0, 44, 90, 179, 226, 271, 316, 329, 362, 450 } /* GRADI CHE SI LEGGONO SOTTO----these are in "raw" degrees, i.e. when going east past 360 degrees, add 360 degrees*/
  
#define AZIMUTH_CALIBRATION_TO_ARRAY \
  { 0, 45, 90, 180, 225, 270, 315, 330, 360, 448 }  /* GRADI REALI

Ciao
Ahh... che rotore usi?   Io il G-1000DXC e non riesco a fare bene il rallentamento.....

IW8ELN-4U, 2022-09-21:
HAM IV