k3ng/k3ng_rotator_controller #103: Sketch too big - text section exceeds available space in board

OZ1SEJ, 2023-03-28:
When using Arduino IDE 2.0.4 to compile for Arduino Uno, I get this compilation error:
```
Sketch uses 37572 bytes (116%) of program storage space. Maximum is 32256 bytes.
Global variables use 1946 bytes (95%) of dynamic memory, leaving 102 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board
```
Visiting [the URL](https://support.arduino.cc/hc/en-us/articles/360013825179) doesn't help much - the suggested fixes seem to involve quite a bit of code refactoring, and I'm afraid to start messing around too much in here.

It was my impression that this should be able to run on an Arduino Uno. Did I misunderstand this?

k3ng, 2023-03-28:
Comment out this line:

https://github.com/k3ng/k3ng_rotator_controller/blob/master/k3ng_rotator_controller/rotator_debug_log_activation.h#L7

( #define DEBUG_DUMP )

...and recompile...

OZ1SEJ, 2023-03-28:
Thanks! That reduced the size to 83%! Perfect :+1:

closed 2023-03-28 by OZ1SEJ