k3ng/k3ng_rotator_controller #106: Feature request "Go To"

bochen17, 2023-12-18:
Hi, on every page of Nextion display there is a button "Go To" and I can enter the grid data. I would like to have a similar button where I can enter the  desired azitumth.
Is that possible?

VK4GHZ, 2023-12-18:
You haven't mentioned which size Nextion you are referring to, nor who the firmware was written by.
The size of the display limits will limit what features can be included on the available real estate.
If, by chance, you are referring to the Nextion firmware that I (VK4GHZ) provided, then have you considered touching the azimuth reading?
You should be able to enter azimuth.

bochen17, 2023-12-19:
Hi, I use Nextion 5 inch, firmware by VK4GHZ. Yes, by touching azimuth indication I can enter the desired azimuth. Thanks for that.    Could you give  me some feedback about preset potencjometr feature that does not allow using screen button nor pc control? Temat: Re: [k3ng/k3ng_rotator_controller] Feature request "Go To"  (Issue #106)Data: 2023-12-19 0:38Nadawca: "VK4GHZ" ***@***.***&gt;Adresat: "k3ng/k3ng_rotator_controller" ***@***.***&gt;; DW: "bochen17" ***@***.***&gt;; "Author" ***@***.***&gt;; 

You haven't mentioned which size Nextion you are referring to, nor who the firmware was written by.
The size of the display limits will limit what features can be included on the available real estate.
If, by chance, you are referring to the Nextion firmware that I (VK4GHZ) provided, then have you considered touching the azimuth reading?
You should be able to enter azimuth.

VK4GHZ, 2023-12-19:
Questions should be asked here:
[https://groups.io/g/radioartisan/topics](url)

bochen17, 2023-12-23:
Hi I managed to change the go to button function in Nextion editor by editing the following data for "Go To" button.
DE=69
ABCKeyboard.tDETitle.txt="Enter Grid:"
ABCKeyboard.inputTxt.txt=""
page DataEntry