k3ng/k3ng_rotator_controller #107: Problem Integrating GY-NEO-6M with K3NG Rotator Controller

gdmcintire, 2024-02-11:
I'm building up a K3NG rotator controller on a Mega2560.   I've successfully integrated a relay shield, az and el incremental encoders, and a Nextion 3.5 inch HMI.  Now I am trying to integrate a GY-NEO6MV2 GPS.   First, I implemented a sketch on an UNO using the Tiny GPS++ library.  It works fine on the UNO...sync, using 11 satellites, correct position, etc.   The K3NG sketch doesn't seem to be processing the GPS data correctly  however.  The display still shows the HL59 as the grid.  I assigned pin 53 to GPS_SYNC and it does not indicate sync with the K3NG sketch.  I tried substituting the Tiny GPS++ library for the Tiny GPS in the code but it wouldn't compile.   Are others successfully using the GY-NEO6MV2 with the K3NG sketch and if so, what did you have to do to get it to work?

Greg, AA5C

gdmcintire, 2024-04-05:
Got a NEO-8 and it only outputs $GP messages.  $GN and $B messages were confusing the TinyGPS library.  Works now.   No responses on the forum in almost two months.

closed 2024-04-05 by gdmcintire

sv2sih, 2024-04-06:
thank you very much sorry for my english i use translator i will try it and if i have any problem i will contact you again thank you again good afternoonKousaxidis Zisis 735 Απρ