k3ng/k3ng_rotator_controller #109: Data display formatting with 16 x 2 LCD

f6fvy, 2024-06-10:
I'm having data formatting problems with a 16x2 LCD.
Depending on the values of Az and El, the string positions move, and the "°" symbol is displayed or not.

[image]

I want the Az and El labels to always be placed in the same place and the values to be formatted on the right with a ° symbol in all cases. This will always fit in a 16-column row.

I use the following options :

In rotator_features.h

#define OPTION_DISPLAY_STATUS
#define OPTION_DISPLAY_HEADING
#define OPTION_LCD_HEADING_FIELD_FIXED_DECIMAL_PLACE

In rotator_settings.h

#define LCD_COLUMNS 16 // 20 //16
#define LCD_ROWS 2 // 4 //2       // this is automatically set below for HARDWARE_EA4TX_ARS_USB and HARDWARE_M0UPU

#define LCD_HEADING_ROW 2
#define LCD_HEADING_FIELD_SIZE 16 // 20
#define LCD_STATUS_FIELD_SIZE 16 // 20

I think the problem(s) are around lines 5800-6100 of k3ng_rotator_controller.ino but it's too complicated for me ;-)

Using v2023.10.06.2200

f6fvy, 2024-06-13:
I finally got it working by modifying few lines.

Lines 6022 and 6025 : The second argument of dtostrf set to 1 instead of 3.
Lines (6052, 6054), (6077, 6079) and (6135, 6137) : Removal of the test "if ((LCD_COLUMNS > 18)..."

Please note that these mods apply to my hardware setup (LCD 16 x 2) and may fail with other configurations. YMMV.

Baerchen239, 2025-02-16:
Hello f6fvy,
I implemented it exactly the same way as you.
However, the ° symbols are not displayed?

Baerchen239, 2025-02-16:
Thanks for your quick reply.
I missed a small thing... misunderstood it.
Now it works perfectly.

[image]